issus/altium-library #9: Connect to Azure failed

zsserg, 2017-06-01:
Hi!
I've done all the setup through desktop tool as described [here](https://github.com/issus/CelestialADB-Desktop), cloned the repo, but Altium still wouldn't connect to the database (see screenshot attached).
[image]
 Can this be a server-side problem?

issus, 2017-06-01:
Did you just create an account? I had an exception about 50 minutes before your issue was created:

```
New exception at setting database roles on Thursday, June 1, 2017 2:16:08 PM (note: no idea what timezone this is!)

Message: An error was reported while committing a database transaction but it could not be determined whether the transaction succeeded or failed on the database server. See the inner exception and http://go.microsoft.com/fwlink/?LinkId=313468 for more information.

---- INNER EXCEPTION ----

Message: Execution Timeout Expired. The timeout period elapsed prior to completion of the operation or the server is not responding.
```

If that's the case, you dont have a database role, and i'll have to create it manually for you

zsserg, 2017-06-01:
Yes, it was about an hour ago. The connection test tool fails with timeout error:
[image]
I do see connection attempt, but Azure server does not seem to respond:
[image]

zsserg, 2017-06-01:
OK, I seem to partially find the reason - blocked outgoing TCP/1433. If I forward it through remote ssh server (local 1433 -> ssh tunnel -> Azure 1433) I am getting other error message - this time Azure firewall is obviously blocking unknown IP. Wrote to uni it-support to see if it would be possible to unblock outgoing TCP.

issus, 2017-06-01:
Ah yes, Universities love to block outbound SQL connections. Your IT department should be able to help you out.

closed 2017-06-01 by issus

zsserg, 2017-06-06:
Just to confirm - indeed it was uni firewall blocking outgoing TCP/1433. Sorry for bothering. ;)