issus/altium-library #11: New User activation code not sent and error

LGKev, 2017-08-17:
I have attempted to sign up, but cannot see the send button in the window (its below bottom of the window and does not allow for scrolling)
 i just hit tab 2 times and then enter, which takes me to the activation code window. 

No activation email sent so I hit resend. An error has occurred shows, without any email sent.

I checked spam.

[image]

[image]

Also is there a lost username/password? I want to reuse the same email I have been using but forgot my password and get the following error:
[image]

logictom, 2017-09-20:
Tried to sign up today, same issue - "An error has occurred." and I never received an email

Saruniks, 2017-11-25:
same problem

yanw4tanabe, 2017-11-29:
same problem here

issus, 2017-11-29:
Closing as this is an issue for the desktop manager software: https://github.com/issus/CelestialADB-Desktop

I'm about to push a code change up which will enable scaling so it can run on a 768-pixel high screen, however, I won't be pushing a new version just yet.

sanketrghadshi, 2017-12-13:
I am receciving any activation code. Its says username/ email does notvexist

sanketrghadshi, 2017-12-13:
I am not receiving any activation code. It says username/ email not valid

luccasc, 2018-07-04:
Hello, i have same problem. I am not receiving any activation code and It says username/ email not valid.
Help. please.

p-vc, 2018-10-09:
Hello,

Been trying for several days, succeeded in installing the program, completing the form, click register, then end up in the activate account window, but never getting any email.  Resend simply shows a please wait and I end up in the same window again.

Tried with 3 diffrnt emails/names etc, no succes.
Is this still active?

G-S-E, 2019-02-25:
I used gmail.com with success, but some other domains fail.

GHardman, 2019-03-18:
Same problem here. Tried with gmail.com and another email, but no code sent, just the message about email not being valid.

closed 2019-03-30 by issus