issus/altium-library #17: No designator on assembly layer

wdecook, 2018-02-05:
There is no reference designator text on the assembly layer. 

I know this text isn't necessary if Draftsman is used to make the assembly drawing, but I can't get Draftsman to create an assembly drawing in the project that is using this library. Similarly, I have issues with using the project releaser with this library.

issus, 2018-02-05:
There was a bug causing me significant pains for reference designator on
the assembly layer, so I didn't add them when creating the library.
Draftsman is also very buggy - please report any issues with draftsman to
Altium directly, it's the only way the bugs have any chance to get fixed
this decade.

wdecook, 2018-02-05:
I'm in the process of working on the draftsman issue with Altium. Was just wondering if maybe you had some insight.

Are you just using your silk layers for assembly drawings?

issus, 2018-02-05:
No, I use a filter to select all the reference designators and shift them
onto a mechanical layer. Its not ideal, generally I'll use Draftsman where
possible but it does REALLY struggle with some large designs.

wdecook, 2018-02-05:
Ok. Thanks

closed 2018-04-03 by issus