issus/altium-library #19: Looking for someone to assist me

jwags818, 2018-03-05:
I have the database installed and it appears to work as I can place parts from it. But I have several questions. 
 If I want to place a part that is not in this library can I add it and if so how?
 What if its a part in a category not currently in the database? How do I create one?
 I am sure I will have more questions  and if someone would be willing to answer them as they come up via email or phone I would be most grateful.

 Jeff

issus, 2018-04-03:
There currently isnt a way for me to vet users adding parts, so it's not an option. As the database is centralised with well over a thousand users, data integrity is important.

I was hoping to have more than 3% of heavy users donate to the Patreon account to cover costs and give me some time to keep adding parts. However, that hasn't happened so I'm only able to provide minimal support for the library, as I'm rarely able to get the opportunity to work on the database after moving country/job.

closed 2018-04-03 by issus

jwags818, 2018-04-04:
Is there a way to export the database to a csv file ?

Jeff

______________________________________
Jeffrey Wagner
Electro-Mechanical Engineer
Lucidyne Technologies
541-286-4897

From: Mark
Sent: Tuesday, April 3, 2018 4:13 PM
To: issus/altium-library
Cc: jwags818; Author
Subject: Re: [issus/altium-library] Looking for someone to assist me (#19)

There currently isnt a way for me to vet users adding parts, so it's not an option. As the database is centralised with well over a thousand users, data integrity is important.
I was hoping to have more than 3% of heavy users donate to the Patreon account to cover costs and give me some time to keep adding parts. However, that hasn't happened so I'm only able to provide minimal support for the library, as I'm rarely able to get the opportunity to work on the database after moving country/job.