issus/altium-library #20: can't connect to Azure?

akohlsmith, 2018-03-16:
Not sure if this is an issue for this repo or for the windows app repo, but after the UDL file is written, Altium can't connect to anything using it. And since I'm on 64-bit Win7 I can't even open the UDL to see what's in there. The tricks for opening a 32-bit UDL on a 64-bit system don't seem to work either.

Can't I get the host/port and give my auth credentials manually and skip the UDL?  For the time being I just downloaded the MDB and am using local database access, but that's suboptimal for what looks to be a promising replacement/augmentation of Altium's vault.

issus, 2018-03-17:
UDL's are not 32bit or 64bit, it's just a text file.

If you cannot connect - check your organisations/schools firewall - this is
the main culprit for failure to connect.

akohlsmith, 2018-03-19:
It's not a firewall issue. This is home internet (I work for myself) with no strange/goofy firewall.

While the UDL file is text, the internet is full of articles on how to connect to databases using UDL files on 64-bit Windows installations, which is why I suspected that was part of the issue.

Even now, csql.database.windows.net is unreachable (or at least has ping turned off):

$ ping csql.database.windows.net
PING northcentralus1-a.control.database.windows.net (23.98.55.75): 56 data bytes
Request timeout for icmp_seq 0

Traceroute gets me into the windows.net network (104.44.11.x) but not much further, again though that might just have ICMP ECHO turned off.

What do you recommend testing to help me resolve the issue? I tried opening the UDL file, which is what led me down that 32/64bit rabbit hole.

defvs, 2018-03-30:
I can confirm it isn't firewall related. When pinging from outside my network (ex : wormly pinging tool) none of my pings works.

issus, 2018-04-03:
Microsoft, like many companies, dont allow ICMP packets through to their servers - too much abuse, this has been pretty standard practice for large data centres for many many years. A ping is no indication of a valid connection. Try a telnet connection to TCP 1433 to confirm the connection.

If the desktop app is telling you that you have a valid firewall rule, it could be your ISP is proxying the web service requests. Vodafone in the UK do this, they even man in the middle HTTPS requests which is pretty appalling!

The server is up and running, and has been showing very steady usage for the past month with no dropouts... so not sure what to suggest if you have a valid firewall rule. You can also try connecting with SQL Server Management Studio or Visual Studio (free versions of either are fine) - just make sure you specify the database as altium_library rather than leaving it blank (blank WILL give you permissions errors if you get to the database.)

defvs, 2018-04-14:
I've tried everything, from going in the DMZ, to VPN, opening every port, disabling firewall. Still can't connect. My ISP is doing shit or what ? Any idea ?

closed 2019-03-30 by issus