issus/altium-library #21: Cannot open server 'csql'

KWAc, 2018-03-23:
Hello,

I'm trying your open source library. Thank you for this ressource.

However, I have the following error when I try test the connection to the Azure server :
Cannot open server 'csql' requested by the login. Client with IP Adress 'xxxxxxxx' is not allowed to access the server. [...]

I already downloaded the Celestial Altium DB Deskstop software and it seems that my Azure access is valid.

Do you have a solution for this error ?

Thank you,

Cyril

issus, 2018-04-03:
Are you using a proxy server on your computer? The server will only add a firewall rule for the IP address the webservices are called from, so if you have an external proxy that will be the IP address you're requesting to add.

closed 2018-05-15 by issus

RadicalRedemption, 2019-05-10:
I am having the same issue, not using any proxy.