issus/altium-library #22: Please specify exactly what MSSQL software is required and how to install it

foobarbecue, 2018-04-09:
I got an email with:

If you have connection issues, then you may be missing the MSSQL driver on your computer, you can download it here:
https://www.microsoft.com/en-us/download/details.aspx?id=29065 

So I went to that page and clicked "Download." The resulting download, bizarrely, was just a confusing text file with, like, country road driving directions to he download, and a note saying "please be certain to pick the 
appropriate component architecture for download."

I'd... like to be certain!? Either I need a better link, or more specific requirements info.

issus, 2018-04-10:
Consider reading the wiki: https://github.com/issus/altium-library/wiki

closed 2018-04-11 by issus

foobarbecue, 2018-04-14:
Looks like that is the necessary information. Would be great if the confirmation email (subject: Welcome to the Celestial Altium Databse) linked to that wiki page instead of the Microsoft page.

zbrozek, 2018-08-24:
Though it would be great to have a HOWTO for a local installation, given that Azure seems to have less-than-stellar uptime.