issus/altium-library #26: Error " Microsoft SQL Server Native Client11.0" login failed for user " "

mhd-munzer, 2018-07-23:
I had my library working for some time, but then I forgot my password and tried every single password I ever used, but still not working. Due to the fact that there is no restore password option, I created another account and activated it. The azure software is accepting my credentials but the udl is showing this error

[image]

I created another account after that, but still the same issue, is there anything I can do for that.

vidavidorra, 2018-08-23:
Same problem here at the moment. I'm 100% sure I'm not behind a firewall or anything. When trying to login the desktop app I get the following error (`Data at the root level is invalid. Line 1, position 1.`. Same when trying to make a new account.

[image]

yldzmuhammed, 2018-08-25:
I have same problem with @vidavidorra. Have you solved the problem?

closed 2019-03-30 by issus