issus/altium-library #34: Use without Celestial ADB Desktop

ThomasW69, 2019-01-22:
Can I use this library without CelestialADBDesktop?

vidavidorra, 2019-01-23:
As an ocasional user of this lib, I can say that it is basically not possible to use it without the desktop.
If you
1. Already have an account to the celestial database
2. Have your current IP approved in the database firewall
you could use the database without the desktop app.

You'd need the desktop app to sign-up an account and approve your IP in the database firewall. Normally I only open the desktop app briefly when I need to fix the firewall. It tends to occasionally expire or something.

ThomasW69, 2019-02-01:
And that's where my problem is. I can't register with Celestal ADB Desktop. I have tried it several times for weeks, but never get a registration email. I've already made a request on the github page of the desktop, but even this hasn't been answered for days.  Is the project still alive at all?

vidavidorra, 2019-02-01:
@ThomasW69 THB, it doesn't seem to be. And if you can't create an account, unfortunately you'll be unable to use this. I'm working on a similar project on the side, but that is not nearly far enough in development to put it publicly on GitHub. I'm planning to release it somewhere around July this year, but no promises. I'm also planning on having the project setup in such a way that 
1. other people can collaborate parts
2. there is no need for a database account (I'm thinking to setup the DbLib to use Excel DB, which can be diff'ed using git or some other tool), this would mean that it can be used offline.

ddavidebor, 2019-02-16:
@vidavidorra Honestly, so many people try to build their library system, me included. Please try contributing to this one instead. Less glory but more utility

amtlib-dot-dll, 2019-02-25:
You can install it in some other PC, and then craft a local database following my steps here https://github.com/issus/altium-library/issues/7#issuecomment-466867608

issus, 2019-03-30:
The project is not dead, I've just been doing 120+hr weeks without much internet access the past year so haven't been able to deal with this much. I'm going to be done with that pretty shortly, and will be working on the database again. I've got a new website and management system coming in a few weeks that will do away with the PC software. Keep on eye on Patreon for updates - https://www.patreon.com/celestialdb.

At the moment, local replication isn't supported but is on the todo list. It's pretty difficult when I'm actively working on the database for users to keep their local database in sync, as there certainly been days when i've added hundreds of components in a day....

closed 2019-03-30 by issus