issus/altium-library #35: Celestial Library Vastly Slowing Altium Down

airexes, 2019-01-29:
Some days when using the library Altium will become unusable. If I even try to move a component in the schematic Altium will lock up for about a minute. Even after restarting my pc the issue is intermittent. Sometimes it immediately returns after a fresh restart. When I close the library the issue disappears. Don't know what else too add that could help.

 I have a 4790k, 32gb ram, gtx 980 ti and everything is on an ssd.

ddavidebor, 2019-02-16:
Hi, try:
* disable autosave
* disable svn function
* closing bomdocs

What version of altium are you on?

lollokara, 2019-02-24:
> Still having this issue...any suggestions are welcome.

use local installation. See issue 7

issus, 2019-03-30:
I don't really see this as a database issue, it shouldn't slow down moving components and such, just browsing/placing components into a schematic, or adding them to a PCB from the schematic. I'd suggest you contact your Altium rep and report it as a bug, they may have a workaround.

closed 2019-03-30 by issus

mohannad999, 2022-04-27:
I face the same problem ... how I can fix that?

engralimirza, 2024-06-07:
> > Still having this issue...any suggestions are welcome.
> 
> use local installation. See issue 7

How to use local installation?

engralimirza, 2024-06-13:
> > > > Still having this issue...any suggestions are welcome.
> > 
> > 
> > > 
> > 
> > 
> > > use local installation. See issue 7
> > 
> > 
> > How to use local installation?
> 
> Please see issue 7, there is a python script to run to have the local lib. It's working just tested.
> 
> Thanks!

Issue 7 is deleted.