issus/altium-library #38: Database connection error - Access Database Engine x64 already installed

patxi-dpg, 2019-08-20:
Hi!!

I'm trying to install de database following the steps of "http://altiumlibrary.com/GetStarted/ConfigureAltium", but Altium gives me the next error:

[image]

I installed the Access engine from microsoft web, but i can't get a sucessfull connection.

Any idea to fix it??

Thanks!!

bretonjustin, 2019-08-20:
You need the 64-bit version of the database engine. Go to https://www.microsoft.com/en-us/download/details.aspx?id=54920 and select the **AccessDatabaseEngine_X64.exe** download.

patxi-dpg, 2019-08-20:
Thanks @bretonjustin; I've already downloaded this file and installed it.

I restarted the PC  again (twice or three times...) and now the error mensage is diferent; now is "Connection failed. Check your connection settings" and I check the settings and user & passwor are correct...

patxi-dpg, 2019-08-20:
MAGIC!! Now all the errors disappeared, and I don't change anything.... and it works!!

Well, thanks for all and your advice @bretonjustin

closed 2019-08-20 by patxi-dpg

64bit-io, 2020-02-11:
Just adding this here, I managed to fix the errors by installing sqlncli.msi (see link below). For some weird reason I needed to restart AD multiple times to get it work.

http://www.microsoft.com/en-ph/download/details.aspx?id=50402

Altium Designer 18
Microsoft Office 2019 64bit
Did not need to have access database engine 64bit installed (I actually installed this, but still had the same error as OP).

Hope this helps.

tomas1808, 2020-02-13:
You are a genius @64bit-io! After weeks trying, that installer fixed everything!

GetzMikalsen, 2020-06-30:
> Just adding this here, I managed to fix the errors by installing sqlncli.msi (see link below). For some weird reason I needed to restart AD multiple times to get it work. [...]

This should be added to the Wiki or FAQ