issus/altium-library #39: Server is down?

hawkenlee, 2019-09-26:
Is this project still running? I cannot contact with the server. How could I use the lib?

hawkenlee, 2019-09-26:
[image]
My error!

sasschary, 2019-09-26:
Issue seconded. I have been unable to connect, and the library website is showing the default Apache website rather than the normal Altium library website.

mojtabashahbandari, 2019-10-07:
The online database will be up again?

mojtabashahbandari, 2019-10-07:
Can anyone tell me what happened to the project server and database?

gyohng, 2019-10-07:
The server crashed and there were no backups. Mark promised to publish the last downloaded database on github at some point. If you need it quicker, please ask on Discord.

closed 2020-02-13 by issus