issus/altium-library #46: Mouser Supplier Link

mumairanwar92, 2020-02-13:
Hi, I couldn't find the supplier information except for digikey. Is there any help you guys can provide so I can add mouser entries in the database for the components I'm using?

issus, 2020-02-13:
We're still working on the functionality for additional suppliers, and hope to have that implemented by the end of the month

closed 2020-02-13 by issus

mumairanwar92, 2020-03-10:
> We're still working on the functionality for additional suppliers, and hope to have that implemented by the end of the month

Hi, sorry for bothering again. Is there any update for the mouser supplier links ?

issus, 2020-03-17:
Hi, sorry for the late reply, i've been moving house.

We're currently just starting beta testing of mouser data collection. It's going to be a few more weeks of work.

aXe-88, 2022-10-02:
Hi, so how do other supplier links work?