issus/altium-library #47: Removed from discord server twice without any explanation or reason

trollcop, 2020-02-25:
For the second time in a month I've been removed from the discord server without any explanation.
Is this an automatic removal for some activity (I didn't do anything retarded to the best of my knowledge)?

I don't remember the removal context last time, but this time I posted a link to a specific discord message (on this same server) which was linking to ECAD/MCAD contribution guide. Shortly after I was no longer in the server.

So uh, what?

issus, 2020-03-04:
Your first removal was with the message "the server is for positive, helpful discussion and your constant negativity and disparaging attitude to less experienced users is not acceptable."

You were banned with the message "Sick of negative attitude and unhelpfulness or disdain shown to new members to the community." after you replied to someone interested in contributing with blatant falsehoods that disparaged my previous comments in regards to interest in community contribution. I will not have someone who has previously been complained about by community members twisting what I have said to reflect negatively upon my interest or commitment to this project, I have put many years of effort, and enough money to buy a house into this project with nothing in return.

Your negative attitude, especially towards students and less experienced engineers (or makers) had been bought to my attention in the past multiple times by community members, hence the initial kick. Once you returned, this did not abate, and the comments you made about my interest in contributors and how I run the library were the last straw. 

We have several student members in the discord server who highly value the positive community that helps them learn and understand electronics and how to improve their design. The other admins and I wish to foster this positive community as a place that people can safely ask questions of more experienced engineers without derision or mocking over their lack of knowledge and/or experience.

closed 2020-03-04 by issus

trollcop, 2020-03-05:
Oh, thanks for the explanation. No idea how discord works so i never seen
those removal messages. All I experienced was removal without any reason.

I wouldn't have rejoined after the first one if I've seen it.

Re: contributions, okay. When i asked to pin the contribution guide, your
wording was pretty clear that you didn't want any randos contributing, so i
just passed the message along. You've literally told me the guides were
written for specific paid people and had no interest in linking them from
the github page or pinning in discord.
Maybe you should work on your English if that's not what you meant.

Anyway, good luck running your safe space.

Will not be returning even if the ban is removed.

Have a nice day.