issus/altium-library #49: Clone the SQL Library

JonasFrey96, 2020-03-24:
Hello, 
is it possible to clone the SQL library since our university is not able to open the ports to accesss the library within the university network. 

Best regards, 

Jonas Frey 
ETH Zurich

tomas1808, 2020-03-25:
Yep, [check this comment by gyohng](https://github.com/issus/altium-library/issues/7#issuecomment-525889298). Worked for me!

XiaoSenLuo, 2020-04-03:
> Yep, [check this comment by gyohng](https://github.com/issus/altium-library/issues/7#issuecomment-525889298). Worked for me!

hello, what the version of python you use that work well? my version is python 3.7, it is not work in windows

issus, 2020-04-03:
The library now has the ability to connect over port 80 rather than 1433 for those in such a situation. Go to the portal and download the DbLib:80 from your logins screen.

closed 2020-04-03 by issus