issus/altium-library #56: Since I pulled the latest update, it took a long time to open Altium.

feijie825, 2020-04-11:
Since I pulled the latest update, it took a long time to open Altium. And Altium always loses response. Has anyone else encountered this problem?

emileakbarzadeh, 2020-04-14:
I am encountering the same issue, where after installing and using components from the library, altium has become extremely laggy. My internet may also be to blame here, since it is extremely slow, however even placing wires in schematic editor has slowed down heavily.

XiaoSenLuo, 2020-04-30:
so do I, I have removed this library from altium designer, because when I install this library, the X2.exe is not response when startup it.

stawiski, 2020-05-01:
Same here! It takes long to open but eventually opens for me. Uninstalling the library speeds things up instantly.

issus, 2020-05-01:
I strong suggest everyone to contact Altium support or post into Altium's BugCrunch. Looking through usage stats and server performance, it's not on my end, this is just Altium.

We're working on an altium extension to replace the components panel, which should help with issues like this and other Altium performance issues.

closed 2020-05-01 by issus