issus/altium-library #58: Is there an automated way of generating the symbol/footprint images?

sargsyanga, 2020-04-16:
Hi,

I am new to this library and still exploring the capabilities, however I intend keeping the symbol/footprint library in sync with the images in PNG format.

Is there an easy/automated way for generating the image of the symbols/footprints after they are added to the library?

I typical scenario would be adding bunch of new components to the library and running a script/tool to generate the images for the new components in "images" folder.

Thank you in advance

currentleak, 2020-04-16:
in Altium, with library opened... select Report-Librairy Report

issus, 2020-04-16:
See my AltiumSharp project, this is what I use to generate the images in the library automatically.

https://github.com/issus/AltiumSharp

closed 2020-04-16 by issus