issus/altium-library #62: Celestial Library Slowing Altium Down

thiagossrs, 2020-07-03:
Hi,

After i installed the Celestial Library my Altium became extremely slow in some operations. Like, change a component footprint, update the PCB from the schematic.

After a lot of research i figure it out why.

When certain operations are performed, the library itself open a connection to the database and execute several query's (more the 10k).

The problem is that from my location the latency to the DB server (db.altiumlibrary.com) is huge (200 ms).

For me this could be solved, by making the database locally or with a closer mirror to my location. But i could not find the DB data in github, not connect to the database server to dump the tables.

Anyone else with the same problem?

Can someone provide me a database backup? So I can use a local server to confirm if this fix the issue or not.

Thanks

thiagossrs, 2020-07-05:
I managed to download the database (i was not setting the default database in the SSMS connection). And i can confirm that running the database locally fix the issue.

If anyone need an updated database, just ask me and I'll upload the database backup that i just cloned.

dadabarghi, 2020-07-08:
Hi thiago
I have the same problem here... But I could not download the database in SSMS even by setting the default database name.
Could you please help me? 

Regards.

thiagossrs, 2020-07-09:
Sure,

Here is the backup.

https://github.com/thiagossrs/altium-library-database

Let me know if you need further assistance.

Best Regards,
Thiago

thiagossrs, 2020-07-09:
Here is the database just in case someone need it: https://github.com/thiagossrs/altium-library-database

dadabarghi, 2020-07-10:
Thank you thiago...
Your help is really appreciated.

issus, 2020-07-13:
@thiagossrs Please dont, your database dump is already out of date :(

closed 2020-07-13 by issus