issus/altium-library #70: Resetting the dblib file causes ODBC error

kyuhyong, 2020-08-27:
I am using altium 17 and was able to add the library successfully and no problem so far. 
However my disk is getting full and wanted to change the library folder so I removed the dblib file in the "Available Libraries" menu as below
[image]
And then copy everything in the altium-library folder into different location and install the dblib file again.
But after that, I am only getting these message saying 
"EOleException error executing:
Select * From [Whatever parts]
Cannot find supplier. May not installed properly"
as below
[image]

It seems like Altium is still trying to find the old DB instead of new one. How can I solve this issue?
Thanks,

issus, 2020-09-02:
Have you tried restarting Altium?

kyuhyong, 2020-09-02:
> Have you tried restarting Altium?

Yes I did but no difference.

issus, 2020-09-05:
Altium issues like this should be reported to Altium, use Bug Crunch (https://bugcrunch.live.altium.com/) or contact support - unfortunately there's nothing we can do as it was working previously... so I don't have any suggestions.

closed 2020-09-05 by issus