issus/altium-library #71: Wrong value in two Ceramic Caps

ing-kuczmik, 2020-08-28:
CAP CER 1UF 35V X5R 0402
C1005X5R1V105K050BC

and

CAP CER 1UF 35V X5R 0402
C1005X5R1V105M050BC

Has some mm as value in the Parameters.

[image]

issus, 2020-08-28:
This was reported on discord after the new version of the crawler was run.
We believe we have found the issue and are going to we if we can get it
fixed today!

issus, 2020-08-29:
The height/thickness issue has been resolved, but it looks as though most caps which were erroneous now do not have a capacitance value... so we're still looking into it.

issus, 2020-09-02:
Capacitance and Voltage values back to what they should be!

closed 2020-09-02 by issus