issus/altium-library #76: 3d step file on wrong layers for

adamschoenwald, 2020-10-09:
Unless otherwise listed below, all chip resistor pcblibs have 3d step file on layer 13 alongside the body outline.
Some 3d models are not centered over the pads.

VISHAY WSHP2818 --> body on 1, step on 13

These files did not have the above issue:
PCB - RESISTOR - CHIP - STACKPOLE CSR0402.PCBLIB
PCB - RESISTOR - CHIP - RES 2512_6432.PCBLIB
PCB - RESISTOR - CHIP - RES 1210_3225.PCBLIB
PCB - RESISTOR - CHIP - RES 0402_1005.PCBLIB

issus, 2020-10-11:
Do you want to submit a pull request for the fixes? Otherwise I'm planning to write a program which will go through the library with AltiumSharp to look for issues like this on historic footprints which pre-date the current guidance :)

adamschoenwald, 2020-10-12:
Yeah I will do one later today.