issus/altium-library #83: Offline Version of database

gaetano83, 2020-12-04:
Is it possible to have an offline version of the database?
Firewall issues and lack of internet connection prevent many users from using the library.

jlublin, 2020-12-15:
I have tested today and it is possible.
I performed this test by writing a script which simply cloned the tables needed from the database and stored it into a local SQLite3 database. However the local database is currently 117 MiB and since that could amount to a lot of data transfers if run often I'm keeping this script for myself until given an OK by the project owner (or maybe a better solution was if a daily or weekly file could be made available by the project)

In the mean time you can try the following snapshot (zip ~10 MiB).
[celestial-2020-12-15.sqlite3.zip](https://github.com/issus/altium-library/files/5696845/celestial-2020-12-15.sqlite3.zip)

Instructions:
1. Install the SQLite3 ODBC driver from http://www.ch-werner.de/sqliteodbc/
2. In Altium, create a new Database Library
3. Choose "Use connection string"
4. Use the connection string from https://www.connectionstrings.com/sqlite3-odbc-driver/ but replace the "Database" path with the path to your SQLite3 DB file
5. Connect and save the .DbLib inside the altium-library/-directory.
6. Now you should be able to add it to your libraries.

ing-kuczmik, 2020-12-15:
I find it a good direction to offer a backup. I don't want to hope that the database will still exist in 3 years.

TheRealHumaniac, 2021-01-02:
This is a great idea. I appreciate your work very much.
The offline database created according to your instructions works flawlessly!

Is it possible that the snapshot DB is not quite complete?
E.g. I did not find the "Antenna" or the "Connector - Amphenol" parts.
This is not a big deal, but if you could provide a complete library, that would be great.

jlublin, 2021-01-10:
> This is a great idea. I appreciate your work very much.
> The offline database created according to your instructions works flawlessly!
>
> Is it possible that the snapshot DB is not quite complete?
> E.g. [...]

I had a look at portal.altiumlibrary.com (db: "altium_library") and i see the same tables, or views actually, as I have in the sqlite database. There is however a table called "RF - Antennas", is that the one you were looking for?
Also I checked and there are no new updates to the database so "celestial-2020-12-15.sqlite3.zip" is still up to date.

MrZANE42, 2021-02-05:
Regarding offline version of the database it would great if the users (us) would get "SHOW VIEW" privileges. That would mean that we wouldn't have to redo the views our self when creating an offline clone.

MrZANE42, 2021-05-11:
Is there any of the contributors/creators reading these issues?
Would very much like to know if it's possible to get the "sql create" lines for the views, or if the "SHOW VIEW" privilege could be added to standard users.
This would saves some time instead of having to reverse engineer them in order to create a offline synced version

adityadrs, 2021-10-15:
@jlublin  can you check if there is any update to the library? I am looking to deploy it locally and would really prefer if it is the latest update (the one you have uploaded here is around 10 months old). Much appreciated./

adityadrs, 2022-07-09:
@jlublin any chance you can do that?

mukeshnagal90, 2022-08-17:
@jlublin Can you please provide us latest database to deploy it offline. It would be great support for us. Kindly help out.

mukeshnagal90, 2022-08-17:
@jlublin It will also be great if you can share now script with us to download database in sqllite3 locally. Please share your script.

mikhailmihalkov, 2023-02-27:
Who managed to make it offline?

mikhailmihalkov, 2023-02-28:
I don't really understand why a public library cannot be available on a computer without an internet? In any case, you will have to convert it to integrated in real time...

mikhailmihalkov, 2023-02-28:
In the form that she now does not work normally... Altium freezes strongly when the library is turned on because the server part is on their side

DhamuVkl, 2024-03-06:
Try this! https://github.com/ximtech/altium-migrator.git  &  https://github.com/ximtech/altium-library.git It worked for me now i can load my components in offline(except price and stock info).