issus/altium-library #87: Can not place components on Schematics

jptalledo10, 2021-04-25:
Hi

I followed the Celestial Install guide and I am using Altium 21 and able to add the database file and access several components. However, when I right-click and select Place XXXX component, nothing happens.

Any idea what could be happening?

Thanks

jjxxs, 2021-05-23:
Hi. You have to configure it as described in the installation guide. E.g. point it to the cloned repository: http://altiumlibrary.com/GetStarted/ConfigureAltium

closed 2022-01-02 by issus

seanboe, 2024-07-02:
I have the same problem and am unsure what "point" means - I've set the database file in altium but nothing happens when I right click and place component. Could you clarify?

AkhilPuranik, 2024-08-14:
> Hi
> 
> I followed the Celestial Install guide and I am using Altium 21 and able to add the database file and access several components. However, when I right-click and select Place XXXX component, nothing happens.
> 
> Any idea what could be happening?
> 
> Thanks

I had a similar issue , I did the following to fix it.
- Set the path for cloning the git repository to c:/altium-library 
- pasted downloaded dblib file from server to same C:/altium-library 
- removed all preinstalled dblib and re installed (file based preferences)
- placed component and it worked!

Note: if the .dblib file is located in c:/altium-library and rest of the git cloned files are in C:/altium-lib/<sub_folder>  then AD21 cannot add  components to sheet.

 
 hope this helps!