issus/altium-library #93: download *.DbLib save github repo clone folder but get "connect failed check your connection settings"

jeffreyhu3, 2021-09-29:
Hi guys,

I follow Getting start from https://altiumlibrary.com/GetStarted, and download Celestial Altium Library - altium_library.DbLib from portal, meanwhile save it to folder which clone from github repository. When I open it by Altium Designer, get nothing buy get error "connect failed check your connection settings". Please help me how to fix it.

[image]

jeffreyhu3, 2021-09-29:
Fix this issue, following https://altiumlibrary.com/GetStarted/Troubleshooting item about **Steps to Fix Missing Driver**,  I just install MSSQL Native Client 64 Bit and the connection failed was fixed.

closed 2021-09-29 by jeffreyhu3

tebbo123, 2023-06-25:
I installed the MSSQL native client 64 bit and i still have the same issue . 

- i don't have a an educational ISP
- TCP 1433 is enabled for Altium designer 22 on my firewall and antivirus
- i have enable network connections on Altium preferences 
- When i click on build to test the connection , i get the errors shown below !
[image]
[image]