issus/altium-library #103: 0805 Capacitors have wrong footprint

dookei, 2022-08-26:
Hi There,
the 0805 SMD capacitors have ver less listance between pads ...this will cause shorts when using Cu with 105um.
Would it be possible to had greated distance between pads?

issus, 2023-02-27:
0805 Capacitors footprint is to the manufacturers recommended footprint (yageos i believe)

closed 2023-02-27 by issus

legier, 2023-11-16:
@issus that does not mean it is recommended for every board. Here is a screenshot from a manufacturer datasheet. You have 3 different dimensions based on desired density level. Typically you should be able to route 2 traces between 0805 pads, currently you cannot route even single one.
[image]