issus/altium-library #127: load component slow

ccwanggl, 2023-07-18:
[image]

my PC info as up picture, it will "loading" one min when i switch categories.

mikhailmihalkov, 2023-08-24:
the power of your PC does not greatly affect the speed, since the library is not local, but server-side; if you want speed, convert it to an integrated one

St0RM53, 2024-04-17:
how can you convert it to local one?

St0RM53, 2024-04-18:
Found how to do it, this database paired to the server is slow as f*ck because the server is slow as a turtle. I don't see the point of having to access the damn server when all the schematics and footprints are stored locally. The rest of the data should be updated on github as well and the library should be able to work offline only. 

Put a damn resistor from this library and you want to change the footprint? Click on the damn edit footprint and wait 5-10mins for the dialog box to open. Complete fail.

engralimirza, 2024-06-07:
> Found how to do it, this database paired to the server is slow as f*ck because the server is slow as a turtle. I don't see the point of having to access the damn server when all the schematics and [...]

You installed library locally?

mikhailmihalkov, 2024-06-07:
> how can you convert it to local one?
https://www.altium.com/ru/documentation/altium-designer/database-library-migration-tools?version=17.1#470403aecd3da7cc72f7c5f08e4ab812

engralimirza, 2024-06-07:
> > how can you convert it to local one?
> > https://www.altium.com/ru/documentation/altium-designer/database-library-migration-tools?version=17.1#470403aecd3da7cc72f7c5f08e4ab812

Thanks man, I'll try this method.

engralimirza, 2024-06-13:
> You installed library locally?

I tried this method, but it can't get it complete. During the migration process it stop and have to start again.