issus/altium-library #142: Install Celestial Altium Library Offline

engralimirza, 2024-06-07:
Hi, 
This library is great and almost has every part you need. The only issue is server speed, it takes too long to load and sometimes it froze the altium. I would like to install this library locally how can I do that?

Also, I can't clone the library I'm getting this error. 

"Cloning into 'C:\Users\alias\OneDrive\Desktop\altium-library'...
remote: Enumerating objects: 98046, done.        
remote: Counting objects: 100% (26340/26340), done.        
remote: Compressing objects: 100% (5736/5736), done.        
error: RPC failed; curl 18 transfer closed with outstanding read data remaining
error: 6614 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output"

insaneinvader, 2024-08-16:
Some tips about making the lib offline may be found [here](https://github.com/issus/altium-library/issues?q=is%3Aissue+is%3Aclosed+offline).

DOX-DESONIXS, 2024-12-16:
DOX-DESONIXS wishes you happy days.
This is a problem that we have experienced recently during file upload and cloning operations and has become a nuisance. But after a long effort and research, I managed to solve this problem on my own system.
I am happy to share my solution steps with you:
1. Make sure that your computer or system is up to date.
2. Github pays great attention to the search engine where the file will be downloaded. For this:
1. If you have any extensions or any applications or work that depend on the search engine in your search engine, temporarily disable them and try downloading again.
3. If you have successfully completed the download process, it is recommended that you install the recommended Altium Designer program immediately afterwards. After these processes, I have prevented both healthy installation, freezing and slow working problems on my computer. And now I can carry out my electronic works without any problems and flawlessly.

-> If you did not get a result and encounter the same problem again, please explore this link. It will help you:
link: https://stackoverflow.com/questions/38618885/error-rpc-failed-curl-transfer-closed-with-outstanding-read-data-remaining

DOX-DESONIX WISHES YOU GOOD WORK.