issus/altium-library #144: Not able to get a proper clean installation

rnicolas, 2024-08-07:
It is not possible to use it without installing MS Access, and when install it, Altium seems not to be happy with this and failing many times. Proper way of installation and dependencies should be provided.

insaneinvader, 2024-08-16:
I successfully installed the library today. I do not have MS Office installed and I have not installed MS Access. I had to install MSSQL Native Client 64 Bit as mentioned [here](https://altiumlibrary.com/GetStarted/Troubleshooting) and [here](https://youtu.be/lBvwvbw83iY?t=198). Windows 11, Altium 24.7.2.

rnicolas, 2024-08-16:
@insaneinvader I'm really happy for you, but it's giving me the same error when I try to install the library after installing MSSQL Native Client 64.

closed 2024-09-06 by issus

rnicolas, 2024-09-09:
Problem is still there. @issus, it would be better to close it as it will not be solved.