issus/altium-library #146: Murata GRM31CR71H475KA12L has incorrect footprint in recent updates

NothingButGunpowder, 2024-09-05:
I recently added the GRM31CR71H475KA12L to a design and Altium brought in the new footprint for one device and the old footprint for another device.  The two do not match; I believe the original is correct (I've had boards made with the old footprint).  Also, I don't know if the height of the new one is correct.  Ignore the overlapping silk in the image- this layout isn't complete.

The old footprint was 'CAP 1206_3216-0.8mm', and the new footprint is 'MURATA CAP GRM31M 1206_3216'
 
[image]

issus, 2024-09-05:
Yes, you're right - the footprint was applied too widely - this should only be for the 1.2mm high GMR31M. I'll get the GRM31C's their own footprint today. The old footprint was generic and made to just one (possibly KEMETs) footprint recommendations, we're slowly expanding the capacitor footprints to match each manufacturer's recommendation for each series.

issus, 2024-09-06:
Should be fixed now, if you can check

NothingButGunpowder, 2024-09-06:
Yes, I can confirm the current check-in (c06f0fa47)  footprint looks like it will fit the cap.  Thanks for maintaining this excellent resource!

[image]
[image]

closed 2024-09-06 by issus